Debian/raspi3-image-spec #1: apt-get fails after base system install: complains about expired release file

a3nm, 2017-10-24:
When running `sudo ./vmdb2/vmdb2 --output raspi3.img raspi3.yaml --log raspi3.log`, image creation fails after installing the base system because apt-get complains about an expired release file:

```
2017-10-25 00:34:27 DEBUG STDOUT: b'I: Base system installed successfully.\n'
2017-10-25 00:34:27 INFO Running step: {'chroot': 'root-fs', 'shell': "echo 'deb http://snapshot.debian.org/archive/debian/20171007T213914Z buster main contrib non-free' > /etc/apt/sources.list\napt-get update"}
2017-10-25 00:34:27 INFO chroot /tmp/tmp33d59zd3 to echo 'deb http://snapshot.debian.org/archive/debian/20171007T213914Z buster main contrib non-free' > /etc/apt/sources.list apt-get update
2017-10-25 00:34:27 INFO Exec: ['chroot', '/tmp/tmp33d59zd3', 'sh', '-c', "echo 'deb http://snapshot.debian.org/archive/debian/20171007T213914Z buster main contrib non-free' > /etc/apt/sources.list\napt-get update"]
2017-10-25 00:34:27 DEBUG run external command: [['chroot', '/tmp/tmp33d59zd3', 'sh', '-c', "echo 'deb http://snapshot.debian.org/archive/debian/20171007T213914Z buster main contrib non-free' > /etc/apt/sources.list\napt-get update"]]
2017-10-25 00:34:28 DEBUG STDOUT: b'Get:1 http://snapshot.debian.org/archive/debian/20171007T213914Z buster InRelease [136 kB]\n'
2017-10-25 00:34:29 DEBUG STDOUT: b'Reading package lists...'
2017-10-25 00:34:29 DEBUG STDOUT: b'\n'
2017-10-25 00:34:29 DEBUG STDERR: b'E'
2017-10-25 00:34:29 DEBUG STDERR: b': Release file for http://snapshot.debian.org/archive/debian/20171007T213914Z/dists/buster/InRelease is expired (invalid since 10d 2h 4min 23s). Updates for this repository will not be applied.'
2017-10-25 00:34:29 DEBUG STDERR: b'\n'
2017-10-25 00:34:29 ERROR Command failed: chroot /tmp/tmp33d59zd3 sh -c echo 'deb http://snapshot.debian.org/archive/debian/20171007T213914Z buster main contrib non-free' > /etc/apt/sources.list
apt-get update
b'Get:1 http://snapshot.debian.org/archive/debian/20171007T213914Z buster InRelease [136 kB]\nReading package lists...\n'
b'E: Release file for http://snapshot.debian.org/archive/debian/20171007T213914Z/dists/buster/InRelease is expired (invalid since 10d 2h 4min 23s). Updates for this repository will not be applied.\n'
2017-10-25 00:34:29 ERROR Command failed: chroot /tmp/tmp33d59zd3 sh -c echo 'deb http://snapshot.debian.org/archive/debian/20171007T213914Z buster main contrib non-free' > /etc/apt/sources.list
apt-get update
b'Get:1 http://snapshot.debian.org/archive/debian/20171007T213914Z buster InRelease [136 kB]\nReading package lists...\n'
b'E: Release file for http://snapshot.debian.org/archive/debian/20171007T213914Z/dists/buster/InRelease is expired (invalid since 10d 2h 4min 23s). Updates for this repository will not be applied.\n'
Traceback (most recent call last):
  File "/mnt/fah/mem/omega_home/apps/raspi3-image-spec/vmdb2/vmdb/app.py", line 95, in run_steps_helper
    method(step, self.settings, state)
  File "/mnt/fah/mem/omega_home/apps/raspi3-image-spec/vmdb2/vmdb/plugins/chroot_plugin.py", line 47, in run
    vmdb.runcmd_chroot(mount_point, ['sh', '-c', shell])
  File "/mnt/fah/mem/omega_home/apps/raspi3-image-spec/vmdb2/vmdb/runcmd.py", line 58, in runcmd_chroot
    return runcmd(full_argv, *argvs, **kwargs)
  File "/mnt/fah/mem/omega_home/apps/raspi3-image-spec/vmdb2/vmdb/runcmd.py", line 53, in runcmd
    return cliapp.runcmd(argv, *argvs, **kwargs)
  File "/usr/lib/python3/dist-packages/cliapp/runcmd.py", line 64, in runcmd
    raise cliapp.AppException(msg)
cliapp.app.AppException: Command failed: chroot /tmp/tmp33d59zd3 sh -c echo 'deb http://snapshot.debian.org/archive/debian/20171007T213914Z buster main contrib non-free' > /etc/apt/sources.list
apt-get update
b'Get:1 http://snapshot.debian.org/archive/debian/20171007T213914Z buster InRelease [136 kB]\nReading package lists...\n'
b'E: Release file for http://snapshot.debian.org/archive/debian/20171007T213914Z/dists/buster/InRelease is expired (invalid since 10d 2h 4min 23s). Updates for this repository will not be applied.\n'
2017-10-25 00:34:29 INFO Something went wrong, cleaning up!
```

It's probably possible to fix this by instructing apt-get to ignore the check, but I'm not sure whether this is the best solution.

stapelberg, 2017-10-25:
The ideal solution for this is tracked upstream at https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=763419 — any help with implementing it is very welcome.

Until then, you’ll either need to replace the snapshot URL with an archive URL (but then you might get different image contents, of course) or disable the check as you did.

a3nm, 2017-10-25:
I see, thanks for the pointer! As this probably won't be fixed soon, maybe it'd be worth it to point out the problem in README.md where you explain how to prepare the image? Or maybe add the workaround to disable the check to raspi3.yaml like you did for other upstream bugs?

In any case, thanks for the explanation, and thanks for this project!

stapelberg, 2017-10-25:
I think adding the problem to the README is fair. Would you like to send a pull request for that?

commit 5910bf20 (2017-10-25): document apt-get workaround

Fixes: #1
closed 2017-10-25 by stapelberg