Debian/raspi3-image-spec #5: Document the firewall rules that disallow remote SSH connections?

a3nm, 2017-10-29:
Hi,

I was trying to log into my Raspberry Pi remotely via SSH and it didn't work, and it took me some time that what prevented it were some iptables rules:

```
> sudo iptables -L INPUT
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         
ACCEPT     all  --  127.0.0.0/8          anywhere             /* RFC3330 loopback */
ACCEPT     all  --  10.0.0.0/8           anywhere             /* RFC1918 reserved */
ACCEPT     all  --  172.16.0.0/12        anywhere             /* RFC1918 reserved */
ACCEPT     all  --  192.168.0.0/16       anywhere             /* RFC1918 reserved */
ACCEPT     all  --  169.254.0.0/16       anywhere             /* RFC3927 link-local */
REJECT     tcp  --  anywhere             anywhere             tcp dpt:ssh /* SSH */ reject-with icmp-port-unreachable

> sudo ip6tables -L INPUT
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         
ACCEPT     all      localhost            anywhere             /* RFC3513 loopback */
ACCEPT     all      fc00::/7             anywhere             /* RFC4193 reserved */
ACCEPT     all      fe80::/10            anywhere             /* RFC4291 link-local */
REJECT     tcp      anywhere             anywhere             tcp dpt:ssh /* SSH */ reject-with icmp6-port-unreachable
```

I agree it makes a lot of sense to have these rules, because the initial password is easy to guess. However don't you think it would be a good idea to advertise the existence of these rules to the user, in README.md or in the motd?

stapelberg, 2017-10-29:
Sure. I’m not sure about the best place for documentation about the image, especially once we publish it as an official Debian-supported image. That shouldn’t block us for now, so feel free to send a PR adding documentation where you see fit.

a3nm, 2017-11-12:
Sure. I just did this, sorry about the delay. Please have a look, and let me know if you'd like me to change anything. :)

stapelberg, 2018-01-04:
This has been merged. Closing.

closed 2018-01-04 by stapelberg