Debian/raspi3-image-spec #11: qemu-debootstrap is failing

jcberthon, 2018-03-24:
While trying to generate the image on Ubuntu 17.10 amd64 (my main computer uses Ubuntu and I would like to use Debian ARM64bit on my new Raspberry Pi 3 B+), it fails to find `/usr/share/keyrings/debian-archive-keyring.gpg`.

In order to solve this problem, on Ubuntu, one should install the Debian Archive Keyring package: `sudo apt install debian-archive-keyring`.

But that was not enough, after solving that problem it fails later on (after a few minutes running) with:

```
2018-03-24 23:41:28 INFO Running step: {'qemu-debootstrap': 'buster', 'mirror': 'http://deb.debian.org/debian', 'target': 'root-fs', 'arch': 'a
rm64', 'components': ['main', 'contrib', 'non-free']}
2018-03-24 23:41:28 INFO Qemu-debootstrap buster /tmp/tmpbjhtleuk http://deb.debian.org/debian arm64
2018-03-24 23:41:28 INFO Exec: ['qemu-debootstrap', '--arch', 'arm64', '--variant', '-', '--components', 'main,contrib,non-free', 'buster', '/t
mp/tmpbjhtleuk', 'http://deb.debian.org/debian']
2018-03-24 23:41:28 DEBUG run external command: [['qemu-debootstrap', '--arch', 'arm64', '--variant', '-', '--components', 'main,contrib,non-fr
ee', 'buster', '/tmp/tmpbjhtleuk', 'http://deb.debian.org/debian']]
2018-03-24 23:41:28 DEBUG STDERR: b'I: Running command: debootstrap --arch arm64 --foreign --variant - --components main,contrib,non-free buste
r /tmp/tmpbjhtleuk http://deb.debian.org/debian\n'
2018-03-24 23:41:28 DEBUG STDOUT: b'I: Retrieving InRelease \n'
2018-03-24 23:41:29 DEBUG STDOUT: b'I: Checking Release signature\n'

(...)

2018-03-24 23:43:35 ERROR Command failed: qemu-debootstrap --arch arm64 --variant - --components main,contrib,non-free buster /tmp/tmpbjhtleuk http://deb.debian.org/debian
```

Note that when I simply run this `qemu-debootstrap --arch arm64 --version` I get:

```
I: Running command: debootstrap --arch arm64 --foreign --version
debootstrap 1.0.91ubuntu1.1
cp: '/usr/bin/qemu-aarch64-static' and '/usr/bin/qemu-aarch64-static' are the same file
```

so maybe this is a Ubuntu 17.10 qemu bug...

stapelberg, 2018-03-24:
I haven’t encountered this issue before.

To further isolate the issue (and work around it), you could try building the image in a Debian chroot, container or VM.

jcberthon, 2018-03-25:
I have tried to run it under a Docker container Debian-based image. I had to run it as privileged and sharing the loopback devices, but still `kpartx` did not work within the container. Actually, the `kpartx` command worked when partly when tried manually, e.g. doing `kpartx -asv raspi3.img` work inside the container but not when doing `kpartx -dv raspi3.img`, the conclusion is when called by vmdb2 there are some "cleaning" problem and then the 2nd or 3rd call to kpartx failed.

I will try to set-up a VM for that.

stapelberg, 2018-03-26:
Did commit fa47ee98fb4c2b258fde4bbd6c4e32ad3954cc22 fix your issue by any chance?

jcberthon, 2018-03-26:
I don't know if it is because I restarted my computer just before or if it is your commit (I would think the latter) that fix it, but it did work today (at least no errors are reported).

I need to test the build image now. At least I can see it has installed raspi3-firmware version 1.20180316-3, so my issue #12 is covered (partly at least).

gwolf, 2018-08-27:
I am closing this issue, as it seems to be either magically fixed or unreproducible. Thanks!

closed 2018-08-27 by gwolf