Debian/raspi3-image-spec #12: Raspberry Pi 3 B+ incompatibilities

jcberthon, 2018-03-24:
I was trying to use your build image from January 2018 on my Raspberry Pi 3 B+.

Simply putting the image on a USB stick and trying it did not work. I see the red LED, then the green LED blink, and then the green LED enter a loop state with 4 long pulse followed by 4 short pulse.

I've tried to edit the config.txt file to change the device tree entry to `bcm2837-rpi-3-b-plus.dtb` and duplicated the existing `bcm2837-rpi-3-b.dtb` as `bcm2837-rpi-3-b-plus.dtb` (simply cp to the new name), as [suggested for Fedora 28 beta](https://www.spinics.net/linux/fedora/fedora-arm/msg11942.html). But I end up in the same error.

I wanted to regenerate the image because perhaps the uboot needs an update, but I do not manage to build it on my platform (see #11). Would you have any advice on how to proceed?

stapelberg, 2018-03-24:
You’ll need version 1.20180316-3 or newer of the https://packages.debian.org/sid/raspi3-firmware package. You can simply extract that package and copy the files to the first partition of the image after writing it to an SD card.

Note that while this makes the device boot, there is some (intermittent?) trouble with the network interface, hence I haven’t uploaded a new image yet.

jcberthon, 2018-03-25:
I've seen your issue on the Raspberry Pi firmware project about the LAN device.

Thanks for the tip with the newer firmware!

I will for the time being install Raspbian and wait for this issue to be solved. I would like this RPi to be a DHCP/DNS server, so if the ethernet port is not working great that's a bit of a problem ;-)

stapelberg, 2018-03-26:
Let’s keep this open until we have a new image out which works on the B+

plugwash, 2018-04-05:
My experiance on the b+

I was able to get it to boot by manually updating the firmware files, specifically copying start*.elf fixup*.dat and bootcode.bin from https://github.com/raspberrypi/firmware . I also had to use hdmi_force_hotplug=1 to make it see my monitor but that is an issue with the kvm switch I use.

Unfortunately I was not able to get a working network, the onboard Ethernet wouldn't see a link at all. I then tried an asix based USB hub with Ethernet, that saw the link but I got errors about USB resources when I tried to actually bring it up.

widar, 2018-06-28:
Ethernet works like a charm with 4.18.0-rc2.
BTW, does anyone know why bcm2837-rpi-3-b-plus.dtb only seems to work when renamed to bcm2710-rpi-3-b-plus.dtb? Is that a bootloader quirk?

jlu5, 2018-07-08:
I managed to get Debian working on a RPi 3 B+ after a fair bit of tinkering: https://code.overdrivenetworks.com/blog/2018/07/03/debian-buster-on-a-raspberry-pi-3-model-b-plus/

Here are my general takeaways, which I hope might help anyone else trying to set this up:
- raspberrypi/firmware and the official Linux sources ship different .dtb files, and it seems these shouldn't be mixed (it caused a boot failure for me at least). With a mainline kernel using the .dtb at `arch/arm64/boot/dts/broadcom/bcm2837-rpi-3-b-plus.dtb` in the [kernel source](https://github.com/torvalds/linux/tree/v4.18-rc2/arch/arm64/boot/dts/broadcom) worked, while `bcm2710-rpi-3-b-plus.dtb` from raspberrypi/firmware worked OK with a custom build of the [raspberrypi kernel fork](https://github.com/raspberrypi/linux).
- USB, HDMI video, and Ethernet just about worked out of the box, though I had a minor issue with the default firewall provided in the the image (the -m comment module for iptables didn’t load with my custom kernel for some reason, causing `ifup` to fail).
- HDMI audio required adding `dtparams=audio=on` to config.txt
- For the Wi-Fi driver to load I needed to copy brcmfmac43455-sdio.txt (NVRAM config file) from the [RPi-Distro/firmware-nonfree](https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm) repository into `/lib/firmware/brcm/`, since Debian does not ship it in firmware-brcm80211. Similar bugs are https://bugs.debian.org/844056 and https://bugs.debian.org/797779, which affect other models that basically load the driver in the same way 
- I had issues with Xorg showing only a cursor on `startx` and crashing a few seconds later. dmesg showed me errors like “vc4: Failed to allocate from CMA”, which I worked around by adding `cma=256M` to `cmdline.txt`

In order to run a custom kernel I purged `raspi3-firmware` so that it wouldn't overwrite `config.txt`. This is probably less necessary as 4.18.x is in experimental now, though I've heard that the arm64 build doesn't quite work yet.

chschlue, 2018-07-09:
> raspberrypi/firmware and the official Linux sources ship different .dtb files, and it seems these shouldn't be mixed (it caused a boot failure for me at least). With a mainline kernel using the .dtb [...]

Mine works with the vanilla bcm2837-rpi-3-b-plus.dtb but does not boot with the Raspi-dtb.

> [...] issue with the default firewall provided in the the image (the -m comment module for iptables didn’t load with my custom kernel for some reason, causing ifup to fail).

There is an open pull request to replace iptables in the default install, the current solution appears needlessly convoluted to me anyway.

> HDMI audio required adding dtparams=audio=on to config.txt

Worked out of the box for me.

> For the Wi-Fi driver to load I needed to copy brcmfmac43455-sdio.txt [...]

Same here.

> worked around by adding cma=256M to cmdline.txt

Same. Kodi even seems to need cma=512M, else vc4 crashes after some time.

> This is probably less necessary as 4.18.x is in experimental now, though I've heard that the arm64 build is actually failing atm.

It compiles fine apparently, there just seem to be duplicate modules in -di packages.

chschlue, 2018-07-09:
> In order to run a custom kernel I purged raspi3-firmware so that it wouldn't overwrite config.txt.

The raspi3-firmware package needs to be updated.
``/etc/kernel/postinst.d/raspi3-firmware`` should allow for configuration of cma and such, maybe through ``/etc/default/raspi3-firmware`` or something. Also brcmfmac43430-sdio.txt is in the package source but not in the package.

bioxz, 2018-08-15:
Hey, I've got my Pi 3 B+ today and went directly for Debian arm64. After changing this build to fetch the latest 4.18 kernel from experimental (rc5) I was able to boot and connect via ssh, USB and HDMI video are also working fine.

But what exactly do I have to do to get audio working at all? I'm quite confused where the problem is. Do I need a newer raspy3-firmware package? A different dtb file? Do I just have to modprobe a missing module?

Also I read I have to add dtparams=audio=on to the /boot/config.txt, while I only have a /boot/firmware/config.txt and changes to that file seem to be gone after a second reboot (and audio still isn't working).

```
aplay -l   
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM vc4-hdmi-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

chschlue, 2018-08-16:
Are you lacking HDMI audio or analog audio?
In any case, as long as you are using a Debian kernel, dtparams have nothing to do with it as the accompanying DT blobs don't use parameters.
`/boot/firmware/config.txt` gets rewritten every time you install a kernel image, basically because there isn't much to configure outside of Raspbian.

chschlue, 2018-08-16:
Are you sure this isn't a configuration problem, like PulseAudio blocking ALSA?
VC4 is apparently working correctly, otherwise you wouldn't have HDMI video either, and the same module also manages audio.

bioxz, 2018-08-16:
I don't have pulseaudio installed, just alsa (libasound2, alsa-utils). I can see that some modules for audio are loaded

```
snd_soc_core          180224  1 vc4
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               106496  3 vc4,snd_soc_core,snd_pcm_dmaengine
snd                    81920  3 snd_timer,snd_soc_core,snd_pcm
soundcore              16384  1 snd
```
and there is a device showing as posted above. But when I open alsamixer, there are no controls. It might be a stupid configuration mistake on my side, maybe I forgot to install something that is required? Alsa can see a device, but there doesn't seem to be any output sinks in it.

st-ashcroft, 2018-08-17:
I think it really is an issue with the driver:

# dmesg | grep vc4
[    8.447485] vc4_hdmi 3f902000.hdmi: ASoC: Failed to create component
debugfs directory
[    8.462766] vc4_hdmi 3f902000.hdmi: vc4-hdmi-hifi <-> 3f902000.hdmi
mapping ok
[    8.479599] vc4_hdmi 3f902000.hdmi: ASoC: no DMI vendor name!
[    8.490586] vc4-drm soc:gpu: bound 3f902000.hdmi (ops vc4_hdmi_ops
[vc4])
[    8.500857] vc4-drm soc:gpu: bound 3f806000.vec (ops vc4_vec_ops
[vc4])
[    8.510764] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops
[vc4])
[    8.520592] vc4-drm soc:gpu: bound 3f206000.pixelvalve (ops
vc4_crtc_ops [vc4])
[    8.530974] vc4-drm soc:gpu: bound 3f207000.pixelvalve (ops
vc4_crtc_ops [vc4])
[    8.541256] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops
vc4_crtc_ops [vc4])
[    8.575488] vc4-drm soc:gpu: bound 3fc00000.v3d (ops vc4_v3d_ops
[vc4])
[    8.584670] fb: switching to vc4drmfb from simple
[    8.600493] [drm] Initialized vc4 0.0.0 20140616 for soc:gpu on
minor 0
[    8.675829] vc4-drm soc:gpu: fb0:  frame buffer device
[   57.309661] vc4_hdmi 3f902000.hdmi: ASoC: can't open interface
3f902000.hdmi: -19

That last line will be repated for every attempt to play sound.

bioxz, 2018-09-09:
For those who didn't notice, since a few days the 4.18 kernel went into unstable. Changing my current installation to the linux-image-arm64 meta-package from unstable went fine, the new current package is linux-image-4.18.0-1-arm64 (which is 4.18.6).

For new installations experimental won't be needed anymore, the kernel can be installed via

`apt install -t unstable linux-image-arm64`

Maybe there are some relevant fixes in the new version.

https://tracker.debian.org/news/985488/accepted-linux-4186-1-all-source-into-unstable-unstable/

perezmeyer, 2018-09-21:
FWIW: as bioxz says the kernel landed in unstable. I reverted HEAD and built the image. Boots bug hangs for me in "random: crng init done"

LorenzoAncora, 2018-09-21:
> [...] hangs for me in "random: crng init done"

@perezmeyer happened to me too on a Raspberry Pi 3 B.
I've resolved by editing `config.txt` to match the version of the initial ramdisk with the version of the loaded kernel.

perezmeyer, 2018-09-21:
> I'm trying now building the image with haveged in it.

Or mounting the SD card and do the right foo to install it, but no changes I'm afraid.

perezmeyer, 2018-09-21:
> > [...] hangs for me in "random: crng init done"
> 
> @perezmeyer happened to me too on a Raspberry Pi 3 B.
> I've resolved by editing `config.txt` to match the version of the initial ramdisk with the version of the loaded kernel.

Thanks! But in my case it seems to match what I have in the boot partition:

```
kernel=vmlinuz-4.18.0-1-arm64
initramfs initrd.img-4.18.0-1-arm64
```

LorenzoAncora, 2018-09-21:
> I've resolved by editing config.txt to match the version of the initial ramdisk with the version of the loaded kernel.
>> Thanks! But in my case it seems to match what I have in the boot partition

@perezmeyer  wait an hour and then press ENTER: if it does nothing, you found a known bug. :-(
-  You should still have the previous ramdisk: replace the new version number with the previous one and boot.
- Other Raspberry Pi users have told me about strange waiting on boot that can last up to two hours when you install a new kernel. One possible explanation is that these devices have difficulty collecting entropy and this can cause a strong lag during boot, probably due to the lack of components suitable for collecting random values. One solution is to install `haveged`.

perezmeyer, 2018-09-21:
> > I've resolved by editing config.txt to match the version of the initial ramdisk with the version of the loaded kernel.
> > > Thanks! But in my case it seems to match what I have in the boot partition
> 
> @perezmeyer wait an hour and then press ENTER: if it does nothing, you found a known bug. :-(

Or sends lots of keystrokes and mouse movements. That did not work either.

> * You should still have the previous ramdisk: replace the new version number with the previous one and boot.

No, I don't. Or should I?

> * Other Raspberry Pi users have told me about strange waiting on boot that can last up to two hours when you install a new kernel. One possible explanation is that these devices have difficulty [...]

Which [I did](https://github.com/Debian/raspi3-image-spec/issues/12#issuecomment-423508376). I'm afraid it does not changes anything :-(

LorenzoAncora, 2018-09-21:
>  You should still have the previous ramdisk: replace the new version number with the previous one and boot.
>> No, I don't. Or should I?

**You should** still have them if you have not removed the previous kernel via the package manager (an action you should never do, **you should always have at least two kernels installed**).
Check your `/boot` directory for `vmlinuz` and `initrd` files.

> @perezmeyer wait an hour and then press ENTER: if it does nothing, you found a known bug. :-(
>> Or sends lots of keystrokes and mouse movements. That did not work either.

Useless. The last time it happened to me both keyboard and mouse did not receive current.
Wait an hour and then press ENTER to see if you can get a prompt. If you do not get a prompt you found a known bug and you need to use the previous kernel (+ initrd) to boot and then reconfigure the system.

> Other Raspberry Pi users have told me about strange waiting on boot that can last up to two hours when you install a new kernel. One possible explanation is that these devices have difficulty [...]

Install `haveged`, enable it (if possible via `systemd`) and then wait before rebooting.
Reason: `haveged` needs a lot of time to collect "true randomness".

PS: have you reduced the voltage of the Raspberry Pi? Did you overclock?

perezmeyer, 2018-09-21:
> **You should** still have them if you have not removed the previous kernel via the package manager (an action you should never do, **you should always have at least two kernels installed**).
> Check [...]

Except that [I created the image from scratch](https://github.com/Debian/raspi3-image-spec/issues/12#issuecomment-423502745).
 
> > @perezmeyer wait an hour and then press ENTER: if it does nothing, you found a known bug. :-(
> > > Or sends lots of keystrokes and mouse movements. That did not work either.
> 
> Useless. The last time it happened to me both keyboard and mouse did not receive current.

Oh, that does works for me, I can plug USB devices and see the kernel loading them. I can also hit ctrl+alt+del and "reboot" the Pi. So keyboard is working.

> Wait an hour and then press ENTER to see if you can get a prompt. If you do not get a prompt you found a known bug and you need to use the previous kernel (+ initrd) to boot and then reconfigure the system.

Which I don't have :-) Note that I'm trying to solve the issue by creating a new image, not basing myself in previous versions.

> Install `haveged`, enable it (if possible via `systemd`) and then wait before rebooting.
> Reason: `haveged` needs a lot of time to collect "true randomness".

Not my experience with other ARM based boards, but I'll just try.

> PS: have you reduced the voltage of the Raspberry Pi? Did you overclock?

No, just created the image, loaded in the microSD and booted it.

LorenzoAncora, 2018-09-21:
> Oh, that does works for me, I can plug USB devices and see the kernel loading them. I can also hit ctrl+alt+del and "reboot" the Pi. So keyboard is working.

So you're **luckier** than me! :-)
Use <kbd>Alt</kbd> + <kbd>SysRq</kbd> + <kbd>9-0</kbd> to set the debugging level of the console immediately after the boot (set it to verbose logging) and see what is really going on.
Once the possible error has been identified, edit the boot configuration: pass the `init` parameter to the kernel with the path of a working shell (bash), in order to directly start that shell and be able to check the system status. Then you can export the kernel ring buffer and the status of the device tree so that the kernel maintainers can detect the problem.

> You should still have them if you have not removed the previous kernel via the package manager (an action you should never do, you should always have at least two kernels installed).
> Check your /boot directory for vmlinuz and initrd files.
>> Except that I created the image from scratch.

So you've never been successful before and this is the first time you've tried installing a 64-bit kernel on your Raspberry Pi 3 B+? 
I upgraded from two previous versions on the Raspberry Pi 3 B and everything went well.

> PS: have you reduced the voltage of the Raspberry Pi? Did you overclock?
>> No, **just created the image**, loaded in the microSD and booted it.
>
> Install haveged, enable it (if possible via systemd) and then wait before rebooting.
> Reason: haveged needs a lot of time to collect "true randomness".
>> **Not my experience with other ARM based boards**, but I'll just try.

So you've had similar problems on other boards. Interesting: altered hardware can affect both the firmware and the OS.
Have you **ever** overclocked, downclocked or otherwise altered the hardware operating parameters of the boards that gave you problems (or of the Raspberry Pi B+)? Have you used standard power supplies recommended by the manufacturer?

perezmeyer, 2018-09-21:
> So you're **luckier** than me! :-)

;-)

> Use Alt + SysRq + 9-0 to set the debugging level of the console immediately after the boot (set it to verbose logging) and see what is really going on.
> Once the possible error has been identified, edit the boot configuration: pass the `init` parameter to the kernel with the path of a working shell (bash), in order to directly start that shell and be able to check the system status. Then you can export the kernel ring buffer and the status of the device tree so that the kernel maintainers can detect the problem.

ACK, will do that.

[snip]
> So you've never been successful before and this is the first time you've tried installing a 64-bit kernel on your Raspberry Pi 3 B+?

Right.

[snip]
> > Install haveged, enable it (if possible via systemd) and then wait before rebooting.
> > Reason: haveged needs a lot of time to collect "true randomness".
> > > **Not my experience with other ARM based boards**, but I'll just try.
> 
> So you've had similar problems on other boards.

Right, but definitely not as bad as this one.

> Interesting: altered hardware can affect both the firmware and the OS.
> Have you **ever** overclocked, downclocked or otherwise altered the hardware operating parameters of the boards that gave you problems (or of the Raspberry Pi B+)?

No.

>  Have you used standard power supplies recommended by the manufacturer?

￼Adjustable DC power supplies with current limitation set at the board's specifications levels. I do this as my day job.

LorenzoAncora, 2018-09-21:
>  Install haveged, enable it (if possible via systemd) and then wait before rebooting.
>  Reason: haveged needs a lot of time to collect "true randomness".
>>  Not my experience with other ARM based boards, but I'll just try.
>
> So you've had similar problems on other boards.
>> Right, but definitely not as bad as this one.
>
> Have you used standard power supplies recommended by the manufacturer?
>> **Adjustable DC power supplies** with current limitation set at the board's specifications levels. I do this as my day job.

Even if you are an expert, this is not a good practice. Unfortunately modern computers and SoCs are very sensitive to alterations in the power system.
The use of non-certified power supplies can cause anomalies in computers, including boards. These problems also affect software and can lead to system instability or  corruption of user data. It does not matter if you are thinking of using quality power supplies: if you want to work stably and safely, you should use the hardware certified by the manufacturer.
Follow the [official specification](https://www.raspberrypi.org/magpi/power-supply/) and if possible use the [official power supply](https://www.raspberrypi.org/products/raspberry-pi-universal-power-supply/).
I know it's a nuisance, but before you can diagnose the software, make sure it's not a hardware problem (especially if the hardware is cheap as in this case).

> [...]
>> ACK, will do that.

When you're done, enter the logs on a pastebin and post _links_ here or send them directly to the kernel maintainers. If necessary, they will ask you for more logs or to execute a few more commands.
Be optimistic: **you are doing a great favor to all users of the B + model and luck is favorable to good men**.

<sup>Have a good time. :-)</sup>

BClark09, 2018-10-09:
@perezmeyer, I had the same problem and I was able to get it to boot by changing root in `cmdline.txt` so that:

```
root=/dev/mmcblk0p2
```

I believe this is discussed in: https://github.com/Debian/raspi3-image-spec/pull/33

Hope that helps!

perezmeyer, 2018-10-09:
@BClark09 whatever it was regenerating the image (after reverting bf61a9bcdc1f9991e59a71a33b3d31a0ffe30730) did the trick.

KaliszAd, 2018-10-11:
Will there be a know good (unofficial) image again? The Debian wiki only includes pointers to RaspberryPi 3, not RaspberryPi 3 B+

gwolf, 2018-10-11:
KaliszAd dijo [Thu, Oct 11, 2018 at 11:14:52AM -0700]:
> Will there be a know good (unofficial) image again? The Debian wiki
> only includes pointers to RaspberryPi 3, not RaspberryPi 3 B+

Yes... Yes, there definitively will be. I promised to adopt this, and
have worked a bit, but September and October were quite hard on me. I
will be back on track soon.

As soon as I can produce an image with a working WiFi, I think I'll
push it up.

mbien, 2018-10-12:
thanks to the great instructions i was able to build an image using the kernel from deb buster without much trouble (Pi 3 B+). Everything works fine and it also boots from usb. (i only edited a few things in the yaml, using iptable-persistant etc)

i noticed the mac address is randomly generated on each boot however - which isn't the case when using the raspbian image on my other Pi. I thought i could easily fix this by statically assigning a mac but when i do that the pi isn't reachable.

st-ashcroft, 2018-10-12:
The mac address isn't passed as a kernel param any more.
It is passed via the device tree.

The trick to make this work on the 3 B+ is to make sure you are running at least a 4.18 kernel and copy (change the path to match the kernel you are running):

/usr/lib/linux-image-4.18.0-1-arm64/broadcom/bcm2837-rpi-3-b-plus.dtb

into /boot/firmware

and then edit the device_tree line in config.txt to match i.e.

device_tree=bcm2837-rpi-3-b-plus.dtb

Unfortunately you need to repeat this process if the initrd is updated as config.txt will be overwritten.
This normally only happens during kernel upgrades.

chschlue, 2018-10-12:
> /usr/lib/linux-image-4.18.0-1-arm64/broadcom/bcm2837-rpi-3-b-plus.dtb
> 
> into /boot/firmware
> 
> and then edit the device_tree line in config.txt to match i.e.
> 
> device_tree=bcm2837-rpi-3-b-plus.dtb

You can also just rename it to bcm2710-rpi-3-b-plus.dtb and the bootloader will load it automatically.
A proposed fix to have raspi3-firmware take care of that is here https://salsa.debian.org/debian/raspi3-firmware/merge_requests/2

mbien, 2018-10-12:
is there no way to automate that with a package which would fix that on kernel update?

gwolf, 2018-10-12:
Michael Bien dijo [Fri, Oct 12, 2018 at 08:42:37PM +0000]:
> is there no way to automate that with a package which would fix that on kernel update?

I plan to incorporate it (or something quite similar, at least) into
raspi3-firmware, so that the manual step is no longer required.

mbien, 2018-10-13:
copying the dtb and adding the device_tree property to config.txt worked perfectly and restored the device mac address. Thanks!

st-ashcroft, 2018-10-15:
> You can also just rename it to bcm2710-rpi-3-b-plus.dtb and the bootloader will load it automatically.
> A proposed fix to have raspi3-firmware take care of that is here https://salsa.debian.org/debian/raspi3-firmware/merge_requests/2

That's combined with removing the device_tree line from config.txt completely is probably the best fix.
That way we'll end up with a build which works on all models with no modifications.

chschlue, 2018-10-15:
Yes, that's exactly what my MR is about.

muammar, 2018-11-04:
Hi all. I recently bought a raspberry pi 3 b+. I followed the instructions on the README, and had just to modify this:

` ` `
diff --git a/raspi3.yaml b/raspi3.yaml
index 86e82fb..d563eec 100644
--- a/raspi3.yaml
+++ b/raspi3.yaml
@@ -85,7 +85,7 @@ steps:
       echo 'APT::Default-Release "buster";' > /etc/apt/apt.conf.d/08default-release
       apt-get update
       apt-get -y --no-show-progress -t unstable install raspi3-firmware
-      apt-get -y --no-show-progress -t experimental install linux-image-4.18.0-rc5-arm64-unsigned
+      apt-get -y --no-show-progress -t experimental install linux-image-4.18.0-trunk-arm64-unsigned
 
   - shell: |
       echo "rpi3" > "${ROOT?}/etc/hostname"
` ` ` 

My rpi was able to boot but gets stuck here:

[image]

Is this related to `hdmi_force_hotplug=1` as it was mentioned by @plugwash?

bioxz, 2018-11-04:
* It's super hard to read that post, you may need to reformat it ;)
* There is no need anymore for experimental sources, the 4.18 already went over unstable into testing.
* I was unable to boot 4.18.0-2, while 4.18.0-1 is working fine. That's something you could try. Also make sure that the /boot/firmware/config contains the correct image names, mine got messed up at one point, pointing to different kernel versions.

widar, 2018-11-05:
> @chschlue <https://github.com/chschlue> :
> Just a remark to the proposed fix - but I'm new to the Raspberry Pi, so a
> could be wrong:
> It's the default for the Raspbian bootloader to use the bcm27* device tree
> files, if not overridden by device_tree in config.txt. This is because the
> bcm27* device tree files correspond to their upstream kernel (which is used
> in Raspbian) while the bcm28* device tree files correspond to the
> downstream kernel (which we use in the Debian image).
> For most people, the proposed fix may be a good idea, to get the image
> working "out of the box". But for people using both, upstream and
> downstream kernel with the corresponding device tree, this fix will cause a
> mess. So far you can install both kernels with their device tree alongside
> and point to the right device tree via the config.txt. But after this fix,
> the upstream device tree files will get overwritten. And you can no longer
> distinguish the device tree files (upstream or downstream) by name.

Why would you want to run Debian with a Raspbian kernel or vice versa?

14:24 Uhr:

Kareema0, 2018-11-06:
ATM, I need to run Debian with a Raspbian kernel because of the hardware 
supported. F.ex. my touch display is only working with a Raspbian kernel 
4.14.y using bcmrpi3_defconfig. The problem seems to be device tree 
related, but I don't have much experience in debugging device tree 
issues so far.

bioxz, 2018-11-06:
> Edit 1: @bioxz using 4.18.0-1 makes things go better (thanks for your suggestion). Now, `/boot/firmware/` is empty in my case. When grepping the log file there is this line:
>
> Any idea?

Obviously it shouldn't be empty. Are you sure the partition is mounted? First partition on your SD card should be a small partition that is mounted to /boot/fimware/, the second partition will contain your /.

muammar, 2018-11-06:
> Obviously it shouldn't be empty. Are you sure the partition is mounted? First partition on your SD card should be a small partition that is mounted to /boot/fimware/, the second partition will [...]

I was reading the `/etc/fstab` and other configuration files mentioned in this issue but everything looked well. I tried mounting the partition from the rescue shell that appeared after failing to boot and could not make it work. @bioxz I can see the same warning when building the image. Does that happen to you, too?

Jackfritt, 2018-11-16:
The only change in raspi3.yaml I had to do after cloning is
from 
apt-get -y --no-show-progress -t experimental install linux-image-4.18.0-rc5-arm64-unsigned
to
apt-get -y --no-show-progress -t experimental install linux-image-arm64

Without this change not boot and ERRORS when building with vmdb2 submodule.

HDMI Video output works. Only the LEDs on Network Plug and the Power LEDs dont 
work after booting is done. More things to be tested.

mbien, 2018-11-17:
just a heads up:
if i upgrade the package udev the Pi does not boot anymore from USB. It gets stuck when it tries to mount the raspbiroot partition. Haven't investigated any further, just excluding the package for now.

details:
libudev1/testing 239-11 arm64 [upgradable from: 239-10]
udev/testing 239-11 arm64 [upgradable from: 239-10]

st-ashcroft, 2018-11-30:
The fix to get wifi support working in a 4.18 kernel is to revert the following:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/arch/arm/boot/dts/bcm2835-rpi.dtsi?id=b0c07c5af6d286f3d3b907743998e9d41f6ab042

commit.

I figured this out by using dtc to turn the working dtb files back into dts and then using diff.
I wish I knew more about devicetree and what the original commit was trying to do.

licaon-kter, 2018-12-18:
Is the missing `/sys/devices/system/cpu/cpu*/cpufreq/` folder a known thing?

Even at idle it stays at 60deg C, I'm thinking that it never changes the frequency, but I can't check it.

`cpufreq_powersave` module is loaded.

gwolf, 2018-12-18:
Licaon_Kter dijo [Tue, Dec 18, 2018 at 06:43:06AM -0800]:
> Is the missing `/sys/devices/system/cpu/cpu*/cpufreq/` folder a
> known thing?
> 
> Even at idle it stays at 60deg C, I'm thinking that it never changes
> the frequency, but I can't check it.
> 
> `cpufreq_powersave` module is loaded.

I don't know if it should be available - ARM systems are quite
different from x86 ones. I just checked on my Cubox-i (i.MX6 quad,
ARMv7/ARM32, Debian Stretch, 4.9.0-8, armhf), it also does not present
the mentioned directory.

But, on x86, CPUs often have different running speeds. In the ARM
world, that's not always the case. The Raspberry configures clock
speed as a boot parameter, as "arm_freq" in the boot partition's
config.txt. You can also install cpufrequtils, and use cpufreq-set /
cpufreq-info, but TTBOMK it does not have much of an effect:

root@cubox『18』~# cpufreq-info
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.
analyzing CPU 1:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.
analyzing CPU 2:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.
analyzing CPU 3:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 0.00 ms.
root@cubox『19』~# cpufreq-aperf
CPU doesn't support APERF/MPERF

gwolf, 2018-12-18:
Gunnar Wolf dijo [Tue, Dec 18, 2018 at 02:53:34PM -0600]:
> I don't know if it should be available - ARM systems are quite
> different from x86 ones. I just checked on my Cubox-i (i.MX6 quad,
> ARMv7/ARM32, Debian Stretch, 4.9.0-8, armhf), it also does not present
> the mentioned directory.

Then again, you might want to refer to this page:

    https://haydenjames.io/raspberry-pi-3-overclock/

jlu5, 2018-12-18:
AFAIK cpufreq and thermal have not been upstreamed yet and are only in the RPi kernel right now? https://elinux.org/RPi_Upstreaming#Status

licaon-kter, 2018-12-18:
> I don't know if it should be available - ARM systems are quite different from x86 ones. 

RPi1B w/ Raspbian has it.

@jlu5 Looks like it`s a the bottom of some TODO, if ever: https://lists.infradead.org/pipermail/linux-rpi-kernel/2018-April/

That's unfortunate.

One solution would be to compile their kernel for aarch64 https://github.com/raspberrypi/linux then?

licaon-kter, 2019-01-17:
@mbien 
>  what frequency the Pi3 B+ is running on when nothing is configured

https://www.raspberrypi.org/documentation/hardware/raspberrypi/frequency-management.md

FYI, I'm running the Raspbian kernel + nf_table modules and it's ok, downclocks to 600MHz as needed.

mbien, 2019-01-17:
> @mbien
> 
> > what frequency the Pi3 B+ is running on when nothing is configured
> 
> https://www.raspberrypi.org/documentation/hardware/raspberrypi/frequency-management.md
> 
> FYI, I'm running the Raspbian kernel + nf_table modules and it's ok, downclocks to 600MHz as needed.

well yes. You would need the raspbian kernel for it to be working - thats what i assumed. I am running debian buster + mainline.

mbien, 2019-01-22:
after a few days uptime i am getting the following msg in the kernel log as my USB drive disconnects.
`
Jan 21 03:14:38 kernel: [ 3181.031839] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 7 - ChHltd set, but reason is unknown
Jan 21 03:14:38 kernel: [ 3181.041577] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04600029
Jan 21 08:47:12 kernel: [23134.804744] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 1 - ChHltd set, but reason is unknown
Jan 21 08:47:12 kernel: [23134.814469] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04600029
Jan 21 14:43:44 kernel: [44526.562061] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 0 - ChHltd set, but reason is unknown
Jan 21 14:43:44 kernel: [44526.571787] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04600029
Jan 21 16:51:39 kernel: [52202.131084] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 4 - ChHltd set, but reason is unknown
Jan 21 16:51:39 kernel: [52202.140810] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04600029
`
the USB drive has the OS and swap on it so i don't get any more logs (system halts obviously). Sometimes there is a day between a channel message, but after 4 channels break, its over.

anyone else seeing something like this in their logs?

I assume the dwc2 USB driver is part of the mainline kernel. I checked my old Pi2 with raspbian and it is using a custom "dwc_otg" driver. (however after searching a bit i found that you can apparently use dwc2 too if you set a flag https://raspberrypi.stackexchange.com/questions/77059/what-does-dtoverlay-dwc2-really-do )

uname -a
Linux 4.19.0-1-arm64 #1 SMP Debian 4.19.12-1 (2018-12-22) aarch64 GNU/Linux

schubter, 2019-01-23:
> after a few days uptime i am getting the following msg in the kernel log as my USB drive disconnects.
> 
> Jan 21 03:14:38 kernel: [ 3181.031839] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 7 - ChHltd set, but reason is unknown
> Jan 21 03:14:38 kernel: [ 3181.041577] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04600029
> 
> Jan 21 08:47:12 kernel: [23134.804744] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 1 - ChHltd set, but reason is unknown
> Jan 21 08:47:12 kernel: [23134.814469] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04600029
> 
> Jan 21 14:43:44 kernel: [44526.562061] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 0 - ChHltd set, but reason is unknown
> Jan 21 14:43:44 kernel: [44526.571787] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04600029
> 
> Jan 21 16:51:39 kernel: [52202.131084] dwc2 3f980000.usb: dwc2_hc_chhltd_intr_dma: Channel 4 - ChHltd set, but reason is unknown
> Jan 21 16:51:39 kernel: [52202.140810] dwc2 3f980000.usb: hcint 0x00000002, intsts 0x04600029
> 
> the USB drive has the OS and swap on it so i don't get any more logs (system halts obviously). Sometimes there is a day between a channel message, but after 4 channels break, its over.
> 
> anyone else seeing something like this in their logs?
> 
> I assume the dwc2 USB driver is part of the mainline kernel. I checked my old Pi2 with raspbian and it is using a custom "dwc_otg" driver. (however after searching a bit i found that you can apparently use dwc2 too if you set a flag https://raspberrypi.stackexchange.com/questions/77059/what-does-dtoverlay-dwc2-really-do )
> 
> > uname -a
> > Linux 4.19.0-1-arm64 #1 SMP Debian 4.19.12-1 (2018-12-22) aarch64 GNU/Linux
> 
> edit: fixed formatting

strangle i have the same issue on one pi 3b+ rev 1.3
all together i have 4 raspberry pi 3b+ rev 1.3 running with the same buster image, all booting from usb and are connected via ethernet. 3 are up and running for almost 3 weeks while one of them is having the same dwc2 errors until a deadlock happens.

i've loaded on a different stick an libreelec and this thing runs without issue

ammoniak, 2019-01-24:
> ATM, I need to run Debian with a Raspbian kernel because of the hardware supported. F.ex. my touch display is only working with a Raspbian kernel 4.14.y using bcmrpi3_defconfig. The problem seems to [...]

same here, the touch screen it self works, but I can't get the X11/Xorg configuration to work.

mbien, 2019-02-03:
> strangly i have the same issue on one pi 3b+ rev 1.3
> all together i have 4 raspberry pi 3b+ rev 1.3 running with the same buster image, all booting from usb and are connected via ethernet. 3 are up and running for almost 3 weeks while one of them is having the same dwc2 errors until a deadlock happens.
> 
> i've loaded on a different stick an libreelec and this thing runs without issue

after reading your reply i decided to try a different SSD USB adapter (from a different vendor) but with no luck, it only took a few hours with the new adapter until i got the first two msgs in the kernel log. (Maybe i am unlucky and both vendors use the same hardware but i don't think this is likely - one is externally powered and universal, the other is not)

not quite sure how to proceed from here. As i understand it dwc2 is part of the mainline kernel and raspbian maintains a different driver called dwc_otg. Assuming this issue does not happen on dwc_otg getting it to run with a mainline kernel is probably not trivial. I might have to give up here and just run raspbian :(

schubter, 2019-02-03:
> after reading your reply i decided to try a different SSD USB adapter (from a different vendor) but with no luck, it only took a few hours with the new adapter until i got the first two msgs in the [...]

i recently ordered a new pi and this one runs without issue, a few days ago i've rebuild an image and put it on the pi that was crashing constantly which is now running perfectly fine. not sure what's different other then the build dates. i haven't checked yet if there is a new kernel.

gwolf, 2019-02-06:
Great news — I just built an image with today's Buster, and the wireless works! :-D

OneTinSoldier66, 2019-02-10:
Hello,

I’m using a Raspberry Pi 3 B+ Rev. 1.3. I downloaded and extracted the 20190206(with linux-image-4.19.0-2) image to a microSD card. When I boot it up it seems to get pretty far into the boot up process, but then I get a kernel panic. After sitting there for a few minutes it then spits out a stack(?) trace.

I don’t know what to do about that, but I wanted to let someone know in case it helps.

[15 more replies]

closed 2019-03-11 by gwolf