Debian/raspi3-image-spec #14: WiFi: missing regulatory.db and problematic txpower

mweinelt, 2018-03-28:
On boot we can see that the `regulatory.db` is missing and cannot be loaded. I couldn't find it in any debian package.
```
[   10.197170] platform regulatory.0: firmware: failed to load regulatory.db (-2)
[   10.207175] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
[   10.207200] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[   10.207213] cfg80211: failed to load regulatory.db
```

Nevertheless I can set the regulatory domain through hostapds `country` setting.
```
# iw reg get
global
country DE: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

But apparently it doesn't honor the 20 dBm txpower limit.
```
# iw dev
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr b8:27:eb:bb:b3:d0
		ssid foobar
		type AP
		channel 5 (2432 MHz), width: 20 MHz, center1: 2432 MHz
		txpower 31.00 dBm
```

Setting a txpower limit via `ìw` sets txpower to 0.0 dBm.
```
# iw dev wlan0 set txpower fixed 20
# iw dev
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr b8:27:eb:bb:b3:d0
		ssid foobar
		type AP
		channel 5 (2432 MHz), width: 20 MHz, center1: 2432 MHz
		txpower 0.00 dBm
```

So I revert it back to `iw dev wlan0 set txpower auto` and it's back up at 31 dBm.

What seems to work is setting the txpower through `iwconfig`, which uses ioctls instead of netlink.

```
# iwconfig wlan0 txpower 20
# iw dev
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr b8:27:eb:bb:b3:d0
		ssid foobar
		type AP
		channel 5 (2432 MHz), width: 20 MHz, center1: 2432 MHz
		txpower 20.00 dBm
```

I verified that the signal strength changes considerably between `auto` (31 dBm) and `fixed 20` (0 dBm) and it's not just a visual glitch. Setting the txpower through `iwconfig` works just fine, just not through `iw`.

mweinelt, 2018-04-05:
`regulatory.db` is the signed version of the regulatory domain specification, it's being made available at https://wireless.wiki.kernel.org/en/developers/Regulatory/wireless-regdb.

I run stretch on an APU2c4 with an ath10k wifi card and it does not experience this issue, but I cannot find a regulatory.db on that either. Instead it has a `regulatory.bin` from the `wireless-regdb` package: https://packages.debian.org/sid/all/wireless-regdb/filelist

stapelberg, 2018-04-05:
Thanks for identifying the Debian issue for this. I’ll close the issue now, as the problem lies with the kernel, not our Raspberry Pi image build.

closed 2018-04-05 by stapelberg