Debian/raspi3-image-spec #17: Image assumes mmc location

nicman23, 2018-04-10:
you should probably use labels for fstab and cmdline (and of course name the partitions accordingly)

stapelberg, 2018-04-10:
Not sure I follow. The image has a defined partition table, what’s wrong with making assumptions about those partitions?

nicman23, 2018-04-10:
the assumption is that an sd card will be used. using a usb device such as a usb stick with its block device in `/dev/sda` fails to boot. also, I know RPi has only one mmc device, but it is bad form to use them in fstab and kernel's cmdline.

stapelberg, 2018-04-10:
Okay. Can you send a pull request for this as well please?

nicman23, 2018-04-10:
i am not familiar with yaml that build the image, where can i get some documentation?

stapelberg, 2018-04-10:
We’re using vmdb2. Maybe https://github.com/larswirzenius/vmdb2/blob/master/doc/en/000.mdwn is a good start? Otherwise, feel free to browse https://github.com/larswirzenius/vmdb2.

gwolf, 2018-08-16:
This has been fixed in our HEAD. No images have yet been produced with this, I'm holding off producing an image until I get time to update and test an updated raspi3-firmware that enables wifi on the 3B+.

closed 2018-08-16 by gwolf