Debian/raspi3-image-spec #20: Built my own image; did not resize FS on first boot

oxwivi, 2018-06-05:
```
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 14.9G  0 disk 
|-mmcblk0p1 179:1    0  299M  0 part /boot/firmware
`-mmcblk0p2 179:2    0  1.2G  0 part /
```
What do I do to resize it now, and how do I make future images to auto resize?

chrisnew, 2018-06-05:
To resize it for now manually: You can delete and re-create the partition using fdisk. After that, you can enlarge the ext4 filesystem using resize2fs.

oxwivi, 2018-06-05:
Yeah, Googled and found [instructions](https://elinux.org/RPi_Resize_Flash_Partitions) to do that.

So the image isn't supposed to automatically do the resizing?

gwolf, 2018-07-26:
I have just replaced rpi3-resizerootfs with a new version; to the extent I have tested, this issue should be solved. Feel free to reopen if you still have issues.

closed 2018-07-26 by gwolf