Debian/raspi3-image-spec #30: Missing SPI support

crjp, 2018-07-19:
SPI is not available since SPIDEV is not compiled into the kernel:

config-4.18.0-rc4-arm64:
```
# CONFIG_SPI_SPIDEV is not set
```

gwolf, 2018-07-31:
FWIW, Christian Schlüter has reported this bug in Debian (which is as far as we can go for building this image!):

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=904043

Thanks Christian!

chschlue, 2018-07-31:
FWIW, despite building a kernel with SPIDEV enabled and enabling the SPI devices using [this](https://gist.github.com/chschlue/105d3e7c710634a4499f475dbe5db684) DT-overlay, I haven't gotten it to work yet.

chschlue, 2018-08-02:
[This](https://github.com/chschlue/rpi3-dt-overlays) appears to work, i.e. I can write to and read from the /dev/spidev*

Unfortunately, I can't reach any RPi physically ATM so I could bridge MISO and MOSI for a quick loopback test.

When it comes to device tree, you are fishing in murky waters as much as I am, right?

gwolf, 2018-08-03:
Christian Schlüter dijo [Thu, Aug 02, 2018 at 04:32:42AM -0700]:
> [This](https://github.com/chschlue/rpi3-dt-overlays) appears to
> work, i.e. I can write to and read from the /dev/spidev*
>
> Unfortunately, I can't reach any RPi physically ATM so I could
> bridge MISO [...]

As murky as they can be :-]

chschlue, 2018-08-27:
Apparently fixed in next 4.18-exp

gwolf, 2018-11-26:
It's done in the kernel. I'm sorry for my extended absence from this project, I intend to pick up the slack and get this moving.
I will be working in the next days in a new version, hopefully it won't take long to have it done!
Am, of course, not closing this until it's properly imported and tested.

gwolf, 2019-07-18:
Umh, I am going to close this issue - Its last action from both Christian and me is "I think it is closed", over six months ago.
I am moving over open issues to Salsa. I am going to declare this issue as closed; please open in Salsa if it is not!

closed 2019-07-18 by gwolf