Debian/raspi3-image-spec #32: Unable to build image: ROOT: parameter not set

perezmeyer, 2018-08-21:
While trying to build the current HEAD I've got:

```
2018-08-21 18:11:40 ERROR Command failed: chroot /tmp/tmpgedynjni sh -c sed -i 's/.dev.mmcblk0p2/LABEL=raspiroot/' ${ROOT?}/boot/firmware/cmdline.txt
b''
b'/usr/bin/sh: 1: ROOT: parameter not set\n'
2018-08-21 18:11:40 ERROR Command failed: chroot /tmp/tmpgedynjni sh -c sed -i 's/.dev.mmcblk0p2/LABEL=raspiroot/' ${ROOT?}/boot/firmware/cmdline.txt
b''
b'/usr/bin/sh: 1: ROOT: parameter not set\n'
Traceback (most recent call last):
  File "/home/lisandro/damian/raspberrypi/raspi3-image-spec/vmdb2/vmdb/app.py", line 95, in run_steps_helper
    method(step, self.settings, state)
  File "/home/lisandro/damian/raspberrypi/raspi3-image-spec/vmdb2/vmdb/plugins/chroot_plugin.py", line 47, in run
    vmdb.runcmd_chroot(mount_point, ['sh', '-c', shell])
  File "/home/lisandro/damian/raspberrypi/raspi3-image-spec/vmdb2/vmdb/runcmd.py", line 58, in runcmd_chroot
    return runcmd(full_argv, *argvs, **kwargs)
  File "/home/lisandro/damian/raspberrypi/raspi3-image-spec/vmdb2/vmdb/runcmd.py", line 53, in runcmd
    return cliapp.runcmd(argv, *argvs, **kwargs)
  File "/usr/lib/python3/dist-packages/cliapp/runcmd.py", line 64, in runcmd
    raise cliapp.AppException(msg)
cliapp.app.AppException: Command failed: chroot /tmp/tmpgedynjni sh -c sed -i 's/.dev.mmcblk0p2/LABEL=raspiroot/' ${ROOT?}/boot/firmware/cmdline.txt
b''
b'/usr/bin/sh: 1: ROOT: parameter not set\n'
2018-08-21 18:11:40 INFO Something went wrong, cleaning up!
```

@gwolf I switched back to 68344d79dacc0312084d747fb7160e3604fd4e70 and the build worked perfectly. So I guess it's related to your latests changes.

chschlue, 2018-08-22:
See PR #33

gwolf, 2018-08-27:
Fixed. Thanks!

closed 2018-08-27 by gwolf