Debian/raspi3-image-spec #34: Desktop environment

JoeKays, 2018-08-24:
Hi,

I am trying to install a desktop environment on buster with kern 4.18 RC5 on the 3 B+.
I tried LXDE and XFCE, both with the same result:
After installation I am not able to login anymore, neither through the desktop, nor through tty.
The boot messages just stop at some point and nothing happens.
Any ideas what I have to do differently?

Thanks

kern.log last messages:
 `Jun 22 11:12:04 rpi3 kernel: [   24.870595] PKCS#7 signature not signed with a trusted key
Jun 22 11:12:05 rpi3 kernel: [   25.156754] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Jun 22 11:12:05 rpi3 kernel: [   25.176607] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Aug 24 17:25:17 rpi3 kernel: [   34.571275] logitech-hidpp-device XXXX:XXXX:XXXX.XXXX: HID++ 4.1 device connected.
Aug 24 17:49:23 rpi3 kernel: [ 1480.026519] PKCS#7 signature not signed with a trusted key
Aug 24 17:49:23 rpi3 kernel: [ 1480.047023] fuse init (API version 7.27)
Aug 24 18:08:53 rpi3 kernel: [ 2649.947500] audit: type=1400 audit(1535134133.219:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/man" pid=22319 comm="apparmor_parser"
Aug 24 18:08:53 rpi3 kernel: [ 2649.963797] audit: type=1400 audit(1535134133.219:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="man_filter" pid=22319 comm="apparmor_parser"
Aug 24 18:08:53 rpi3 kernel: [ 2649.979813] audit: type=1400 audit(1535134133.223:4): apparmor="STATUS" operation="profile_load" profile="unconfined" name="man_groff" pid=22319 comm="apparmor_parser"
Aug 24 18:13:03 rpi3 kernel: [ 2900.110325] PKCS#7 signature not signed with a trusted key
Aug 24 18:13:03 rpi3 kernel: [ 2900.125653] PKCS#7 signature not signed with a trusted key
Aug 24 18:13:03 rpi3 kernel: [ 2900.147003] PKCS#7 signature not signed with a trusted key
Aug 24 18:13:03 rpi3 kernel: [ 2900.158302] NET: Registered protocol family 5
Aug 24 18:16:35 rpi3 kernel: [ 3111.866095] audit: type=1400 audit(1535134595.137:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/evince" pid=26103 comm="apparmor_parser"
Aug 24 18:16:35 rpi3 kernel: [ 3111.882539] audit: type=1400 audit(1535134595.141:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/evince//sanitized_helper" pid=26103 comm="apparmor_parser"
Aug 24 18:16:35 rpi3 kernel: [ 3111.900704] audit: type=1400 audit(1535134595.149:7): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/evince-previewer" pid=26103 comm="apparmor_parser"
Aug 24 18:16:35 rpi3 kernel: [ 3111.918068] audit: type=1400 audit(1535134595.153:8): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/evince-previewer//sanitized_helper" pid=26103 comm="apparmor_parser"
Aug 24 18:16:35 rpi3 kernel: [ 3111.937049] audit: type=1400 audit(1535134595.161:9): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/evince-thumbnailer" pid=26103 comm="apparmor_parser"
Aug 24 18:16:35 rpi3 kernel: [ 3111.954539] audit: type=1400 audit(1535134595.165:10): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/evince-thumbnailer//sanitized_helper" pid=26103 comm="apparmor_parser"`

chschlue, 2018-08-27:
Have you tried raising CMA size in `/boot/firmware/cmdline.txt`?

gwolf, 2018-08-27:
FWIW I don't think this is related to this project - The scope of raspi3-image-spec is to produce a minimal working bootable Debian image. If it managed to boot, you logged in, and after installing some packages something is broken... The bugs happen elsewhere in Debian!
I am not *yet*  closing this bug as we might be able to help you pinpoint this down (after all, many among us are Debian people ;-) )
Don't expect to find this in kern.log; if anything, I would suggest you trying to enable the systemd debug shell before installing the desktop environments:

    systemd enable debug-shell
    systemd start debug-shell
That way, you will have in tty9 (Ctrl-Alt-F9, using a HDMI monitor and keyboard) an open root shell no-matter-what. Use it to investigate journalctl or `/var/log/syslog` / `/var/log/auth.log`, as you prefer. Then, we might be able to redirect you to file the bugs accordingly.

JoeKays, 2018-08-28:
@chschlue No, what does that mean?

@gwolf Thank you very much, I am aware, that the scope of this project is different, but I also thought it should be possible to track this down.

I didn't know about the debug-shell trick, this will definitely help me, thanks a lot!

chschlue, 2018-08-28:
JoeKays <notifications@github.com> schrieb am Di., 28. Aug. 2018, 20:13:

> @chschlue <https://github.com/chschlue> No, what does that mean?
>
The VC4 video driver uses CMA (Contiguous memory allocator). The allocated
size can be set with the cma kernel command line option in cmdline.txt. you
might need more than the default 64M.
Note that this parameter is currently overwritten on every kernel update.
This might change if my patches to raspi3-firmware are accepted.

>

oxwivi, 2018-09-06:
How much is allocated to CMA in Raspbian? It'd be pretty easy to figure if that's what we're looking for by replicating their CMA settings.

gwolf, 2018-09-06:
oxwivi dijo [Thu, Sep 06, 2018 at 12:52:46AM -0700]:
> How much is allocated to CMA in Raspbian? It'd be pretty easy to
> figure if that's what we're looking for by replicating their CMA
> settings.

By default, 64M

yazdan, 2019-01-17:
I had a similar issue. I was able to see greeter and login in using tty. After I logged in using greeter I saw just black screen and in some areas of my monitory I could see mouse pointer change to input.

After I investigated journalctl I saw that there is `drm:vc4 _s_create` kernel message. 

I increased `/boot/firmware/cmdline.txt` to 128 and my problems resolved.

I hope this help you resolve the issue

gwolf, 2019-07-18:
The issue seems to be fixed (and has had no activity in six months). I am closing this issue. In case you want to reopen, please do so - but at its new address, in salsa.debian.org — https://salsa.debian.org/raspi-team/image-specs/issues/1

closed 2019-07-18 by gwolf