Debian/raspi3-image-spec #35: Unable to use GPIO

paddatrapper, 2018-08-26:
Using this image, Python and sysfs access to the GPIO pins is unavailable. When using Python3 the following happens:
```
$ sudo python3 bcs.py 
Traceback (most recent call last):
  File "bcs.py", line 18, in <module>
    import RPi.GPIO as GPIO
  File "/usr/lib/python3/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!
```

And using sysfs:
```
$ sudo -i
# echo "4" > /sys/class/gpio/export
-bash: echo: write error: Invalid argument
```

xdissent, 2018-08-27:
They're offset by 458, so "4" should be "462". You can tell by looking at `/sys/kernel/debug/gpio`

jlu5, 2018-08-27:
Re RPi.GPIO not working: from what I can tell this is not a Debian-specific issue. [Quite a few](https://github.com/adafruit/Adafruit_Python_GPIO/issues/84) [RPi libraries](https://github.com/jgarff/rpi_ws281x/issues/289#issuecomment-383919051) try to detect the CPU revision (and by extension, system model) from `/proc/cpuinfo`. But that does not show the required info on 64-bit systems, as reported as https://github.com/raspberrypi/linux/issues/2110.

For Raspberry-GPIO in particular there is this ticket for this problem: https://sourceforge.net/p/raspberry-gpio-python/tickets/161/

paddatrapper, 2018-08-27:
Thanks, I have updated the Debian [wiki page](https://wiki.debian.org/RaspberryPi3) to include this information.

closed 2018-08-27 by paddatrapper