Debian/raspi3-image-spec #36: Configure WiFi the same way as debian-installer

oxwivi, 2018-09-09:
On regular systems, providing WiFi info during installation will automatically let Debian connect when the the installation is done and the system rebooted. This applies to headless non-GUI systems too.

I know couple of ways to make it connect to WiFi over the command line, but I want the new image to connect to WiFi by default instead of having me to connect it to `ssh` or plug in monitors and keyboards to configure WiFi every time.

Basically, I want to know how `debian-installer` does it so I can have it done in advance with the `.yaml` if possible.

gwolf, 2018-09-12:
I don't think we can do the same as d-i, but I was thinking on allowing the user to specify WiFi settings in one of the files in /boot/firmware. I still haven't come up with anything on it, though :-(

oxwivi, 2018-09-12:
Why not? How exactly does the `d-i` do it? Is there any reference doc for it? It's quite annoying because I don't have proper Ethernet switch handy at my desk.

gwolf, 2018-09-12:
Because I expect the most common way to use this image, once a couple of bugs are ironed out and I am able to produce a good-enough one again, is that users download the prepared image. I expect relatively few people to build their own using this infrastructure, so providing the information in the yaml would not work.
I am not involved with `d-i`, but it requires interactivity — Not what this project aims to.
Of course, I am very happy you and others build this image from the repository, but its utility will be maximum when users can just download the built image, flash it to their SD cards, and happily boot.

gwolf, 2019-02-06:
Hi,
Sorry for the long time leaving this hanging. I believe this issue is not relevant to this project in particular, so I'm closing it.

closed 2019-02-06 by gwolf