Debian/raspi3-image-spec #37: Wireless interface disappeared (Raspberry Pi 3 B)

LorenzoAncora, 2018-09-17:
After upgrading from `4.16.0-2-arm64` to `4.18.0-1-arm64`, interface `wlan0` is missing and I've got no wireless connection. Reinstalling the firmware and downgrading to an older kernel does not help.
The Ethernet connection works correctly but I've got no wireless connectivity and no wireless interfaces show up in `lsusb` or `ifconfig`.

LorenzoAncora, 2018-09-20:
@gwolf @chschlue Is this an unsolvable problem? :-)

chschlue, 2018-09-20:
Sry, don't have much time lately.

Apparently, something was screwed up in the 4.18 DT files for rpi3b(+). Unfortunately, those are a PITA to debug because of a dozen levels of includes and such.

I noticed and reported this during the 4.18-rcs, but it seems nobody on earth has looked into it any further yet.

FWIW, I worked around it by copying bcm2837-rpi-3-b*.dtb from the old 4.16 package to /boot/firmware/bcm2710-rpi-3-b*.dtb for the time being.

LorenzoAncora, 2018-09-20:
@chschlue Thank you.
Can you post the file here so I can try if the solution works and also remains as an aid to the rest of the users?

chschlue, 2018-09-20:
https://gist.github.com/chschlue/e0e02ecb1c455a7c8db3a53f9b408d6b

Please give feedback whether this works for you.

LorenzoAncora, 2018-09-20:
**It works**. Thank you.
Adding [bcm2710-rpi-3-b.dtb](https://gist.github.com/chschlue/e0e02ecb1c455a7c8db3a53f9b408d6b/raw/16ec1953f34534a941bb90b2f6f42f30888157fc/bcm2710-rpi-3-b.dtb) to `/boot/firmware` works on _Raspberry Pi 3 B_ with the `brcmfmac` driver.

Step-by-step procedure:
1. (LOCAL): `scp "bcm2710-rpi-3-b.dtb" remoteUser@remoteIP:~/`;
2. `sudo cp ~/bcm2710-rpi-3-b.dtb /boot/firmware/`;
3. append `brcmfmac` to `/etc/modules`;
4. `sudo systemctl reboot now`;
5. `wlan0` should be visibile now (OUTPUT of `sudo iwconfig`).

closed 2018-09-20 by LorenzoAncora

chschlue, 2018-09-20:
Beware of kernel updates. raspi3-firmware will overwrite this with the current (broken) DT blob.

LorenzoAncora, 2018-09-20:
@chschlue 
Temporary solution: `chattr +i /boot/firmware/bcm2710-rpi-3-b.dtb` should prevent accidental overwrite.
Correct solution: a script that automatically downloads and inserts the _dtb_ file in the `/boot/firmware` directory after the kernel update.

Example code in bash script:
```
DTBURL="https://gist.github.com/chschlue/e0e02ecb1c455a7c8db3a53f9b408d6b/raw/16ec1953f34534a941bb90b2f6f42f30888157fc/bcm2710-rpi-3-b.dtb"
DTBDST="/boot/firmware/bcm2710-rpi-3-b.dtb"

if [ ! -f $DTBDST ]; then
        echo "Installing DTB for Raspberry Pi 3 B from" $DTBURL "..."
        wget --quiet --timeout=7 --random-wait --tries=3 --retry-connrefused $DTBURL --output-document=$DTBDST
        if [ -f $DTBDST ]; then
                echo "DTB UPDATED."
        else
                echo "UPDATE FAILED, TRY AGAIN."
        fi
else
        echo "DTB is already installed in" $DTBDST ", backup and delete this file to force an update."
fi
```

I wrote it on the fly but it should be a good starting point. Let me know if I can help you further. :-)

chschlue, 2018-09-20:
Does FAT support an immutable flag?

Anyway, both solutions are temporary. The correct solution would be to fix device tree sources so 4.19 installs accompanied by a working blob again.

Another temporary solution: comment out the relevant lines in ```/etc/kernel/postinst.d/50-raspi3-firmware```

LorenzoAncora, 2018-09-20:
> Anyway, both solutions are temporary. The correct solution would be to fix device tree sources so 4.19 installs accompanied by a working blob again.
> Another temporary solution: comment out the relevant lines in /etc/kernel/postinst.d/50-raspi3-firmware

Correcting the original package is obviously the **best** solution.
**If** you can't get a working blob you can use the script I provided by programming its execution immediately after each kernel update, replacing `DTBURL` and `DTBDST` with the parameters provided as an argument.

Replace `DTBURL=...` and `DTBDST=...` with:
```
DTBURL=$1
DTBDST=$2
```
...and launch the script as: `script.sh "<DOWNLOAD URL>" "<DESTINATION FILE>"`.
It is a little integrated but common and quite versatile system: if desired, with some minor changes it can also verify the integrity of the downloaded _dtb_ (checksum) and make a backup of the existing _dtb_.

> Does FAT support an immutable flag?

No, you'll need to use ext.
This solution only works to keep the file safe out of the boot partition, which (as I understand it) must be FAT to be recognized by Raspberry Pi.

LorenzoAncora, 2018-09-22:
@chschlue  that's no good? :-)