Debian/raspi3-image-spec #39: on which platform to carry out

zd59, 2018-10-26:
HI!

Sorry, but from the text here I can not guess, on which platform this manual must be executed?
Image created is for ARM64, in text are "apt install". Confused: if I'm creating an image from packages installed, this means, that the system is already installed and I'm logged in. If this is not correct, the installed packages (apt install) are not for arm64 and will not work there.
So what existed first, egg or chicken?
Please uncover the above secret.
My posibilities are ARM32 raspbian or Slackware64 on Intel. If needed, I can install Virtualbox Debian machine on Intel platform.

Hardly waiting for the answer.

Best regards,

Zdenko

chschlue, 2018-10-27:
vmdb2 will download and install arm64 packages, your host platform doesn't matter.

zd59, 2018-10-27:
Thank you Christian.

I figured out that myself, when installed Debian on VirtualBox.
Troubles at image creation:

1. creating from GIT snapshots
I used suggestion
"Due to a missing feature
<https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=763419> on snapshots, to
make the build work, you have to disable an expiration check by APT. To do
so, edit raspi3.yaml to replace all apt-get invocations with apt-get -o
Acquire::Check-Valid-Until=false"
But regardless of that I saw that apt-get ... returns sources are expired.
Why
cliapp.app.AppException: Command failed: chroot /tmp/tmp0lbjnh9o apt-get
update
b'Hit:1 http://snapshot.debian.org/archive/debian/20180930T210043Z buster
InRelease\nReading package lists...\n'
b'E: Release file for
http://snapshot.debian.org/archive/debian/20180930T210043Z/dists/buster/InRelease
is expired (invalid since 19d 10h 25min 17s). Updates for this repository
will not be applied.\n'
2018-10-27 08:58:11 INFO Something went wrong, cleaning up!

2. creating  from egular Testing (*buster*) distribution, leave raspi3.yaml
as it is
ERROR: Command failed: chroot /tmp/tmppkxavdv_ sh -c echo 'deb
http://deb.debian.org/debian unstable main contrib non-free' >>
/etc/apt/sources.list
...
...\nSetting up raspi3-firmware (1.20180619-1) ...\nReading package
lists...\nBuilding dependency tree...\nReading state information...\n'
b"debconf: unable to initialize frontend: Dialog\ndebconf: (TERM is not
set, so the dialog frontend is not usable.)\ndebconf: falling back to
frontend: Readline\ndebconf: unable to initialize frontend:
Readline\ndebconf: (Can't locate Term/ReadLine.pm in @INC (you may need to
install the Term::ReadLine module) (@INC contains: /etc/perl
/usr/local/lib/aarch64-linux-gnu/perl/5.26.2 /usr/local/share/perl/5.26.2
/usr/lib/aarch64-linux-gnu/perl5/5.26 /usr/share/perl5
/usr/lib/aarch64-linux-gnu/perl/5.26 /usr/share/perl/5.26
/usr/local/lib/site_perl /usr/lib/aarch64-linux-gnu/perl-base) at
/usr/share/perl5/Debconf/FrontEnd/Readline.pm line 7, <> line 1.)\ndebconf:
falling back to frontend: Teletype\nE: Can not write log (Is /dev/pts
mounted?) - posix_openpt (19: No such device)\n/usr/bin/ls: cannot access
'/boot/vmlinuz-*': No such file or directory\nraspi3-firmware: no kernel
found in /boot/vmlinuz-*, cannot populate /boot/firmware\nE: Unable to
locate package linux-image-4.18.0-rc5-arm64-unsigned\nE: Couldn't find any
package by glob 'linux-image-4.18.0-rc5-arm64-unsigned'\nE: Couldn't find
any package by regex 'linux-image-4.18.0-rc5-arm64-unsigned'\n"
Something went wrong, cleaning up!
Exec: ['umount', '/tmp/tmppkxavdv_/.//boot/firmware']

Please be so kind to help me.

Best regards,

Zdenko Dolar