Debian/raspi3-image-spec #40: Not booting (or booting without network) with linux-image-4.18

mdosch, 2018-10-27:
Thanks for your work I really appreciate to run a "real" Debian on my Pi3. Everything worked smooth for me so far but since I upgraded the kernel to 4.18 it is either not booting or not bringing up the network as I can neither ping the Pi nor SSH to it. Is there any known issue? 
Maybe I find some time to connect it to a TV and keyboard and investigate this issue this weekend. Any hints where to look for the root cause beside the obvious like the syslog?

chschlue, 2018-10-27:
Copper or air?

The DTB accompanying 4.18 unfortunately kills WiFi. A workaround is to use a DTB from earlier kernels.

LorenzoAncora, 2018-10-27:
> The DTB accompanying 4.18 unfortunately kills WiFi

Can be related to #41?

mdosch, 2018-10-27:
Damn, Github tricked me again by searching only for open issues by default...
Copying the old dtb did the trick. Sorry for not being more thorough when searching and thanks again for your great work. :)

closed 2018-10-29 by mdosch