Debian/raspi3-image-spec #41: Wi-Fi Interface disappeared after sudden reboot

LorenzoAncora, 2018-10-27:
### Problem
After entering the DTB files in the firmware directory, everything was fine and the wireless interfaces worked perfectly. Yesterday evening the electric power went out for ~2 hours and, when the board restarted, `wlan0` disappeared. After several reboots the situation is unchanged, even though the driver `brcmfmac` is in memory and is in use.

### Details
- The Ethernet and local interfaces are fully functional, the problem concerns only the wireless interfaces;
- Apparently the kernel ring buffer does not contain any information useful for diagnosing the problem;
- `/boot/firmware` contains only two DTB files, the firmware configuration for boot and some kernel binaries.

<sup>Related to: #37</sup>

LorenzoAncora, 2018-11-01:
Probably I should go back to using Raspbian which does not have these absurd problems.

jlu5, 2018-11-02:
If you're using custom .dtb's, it may help to purge the raspi3-firmware package so that it can't overwrite things (I stumbled into this problem setting up my RPi too)

LorenzoAncora, 2018-11-03:
@jlu5 TY for you suggestion.
> If you're using **custom .dtb's**

No, I am only using older versions of the official ones. Without doing that the kernel does not recognize the wireless interface and I cannot use Internet.

> **purge the raspi3-firmware package** so that it can't overwrite things

This package contains all the proprietary files necessary to boot a Raspberry Pi<sup>&reg;</sup> 3 board.
Purging this package will remove those (proprietary) files:
```
/etc/kernel/postinst.d/raspi3-firmware
/usr/lib/raspi3-firmware/bootcode.bin
/usr/lib/raspi3-firmware/fixup.dat
/usr/lib/raspi3-firmware/fixup_cd.dat
/usr/lib/raspi3-firmware/fixup_db.dat
/usr/lib/raspi3-firmware/fixup_x.dat
/usr/lib/raspi3-firmware/start.elf
/usr/lib/raspi3-firmware/start_cd.elf
/usr/lib/raspi3-firmware/start_db.elf
/usr/lib/raspi3-firmware/start_x.elf
```
On paper it's not a bad idea but **what are the secondary effects**?