Debian/raspi3-image-spec #47: error on image creation

Gru-gru, 2018-12-19:
When running the line "sudo env -i LC_CTYPE=C.UTF-8 PATH="/usr/sbin:/sbin:$PATH" ./vmdb2/vmdb2 --output raspi3.img raspi3.yaml --log raspi3.log", I get:

"ERROR: No runner implements step with keys fs-type, start, mkpart, part-tag, end, device"

Seems to come from the class NoMatchingRunner from vmdb2/vmdb/step_list.py. I tried with an older version (e.g., commit 284f7891cc54da9a9427cea9e108fcc31d15e3bf) and got the same error.

I am on Debian Stretch stable, and I think I installed everything that was necessary.

lpsinger, 2018-12-27:
I'm also getting this error message on Debian buster.

gwolf, 2019-01-15:
This issue has been addressed by @liwfi (pull request #49); I just tested it today under Sid, and works fine.

closed 2019-01-15 by gwolf