Debian/raspi3-image-spec #53: Error on image creation (ERROR expected str, bytes or os.PathLike object, not NoneType)

intilinux, 2019-01-22:
[raspi3.log](https://github.com/Debian/raspi3-image-spec/files/2784142/raspi3.log)

OS: Debian buster, i'm using the original raspi3.yaml available on repository (updated to today, no other mods). The image is created but i noticed that it does not contains "/etc" folder. 

2019-01-22 16:48:11 ERROR expected str, bytes or os.PathLike object, not NoneType
Traceback (most recent call last):
  File "/home/developer/debian_rasp_build/raspi3-image-spec/vmdb2-master/vmdb/app.py", line 107, in run_steps_helper
    method(step, self.settings, state)
  File "/home/developer/debian_rasp_build/raspi3-image-spec/vmdb2-master/vmdb/plugins/mkfs_plugin.py", line 51, in run
    vmdb.runcmd(cmd)
  File "/home/developer/debian_rasp_build/raspi3-image-spec/vmdb2-master/vmdb/runcmd.py", line 53, in runcmd
    return cliapp.runcmd(argv, *argvs, **kwargs)
  File "/usr/lib/python3/dist-packages/cliapp/runcmd.py", line 55, in runcmd
    exit_code, out, err = runcmd_unchecked(argv, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/cliapp/runcmd.py", line 106, in runcmd_unchecked
    kwargs)
  File "/usr/lib/python3/dist-packages/cliapp/runcmd.py", line 143, in _build_pipeline
    stderr=stderr, close_fds=True, **kwargs)
  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1453, in _execute_child
    restore_signals, start_new_session, preexec_fn)
TypeError: expected str, bytes or os.PathLike object, not NoneType

gwolf, 2019-02-06:
I cannot reproduce the issue you describe. Can you provide more insight on your build process?

gwolf, 2019-07-18:
No answer on this for too long. I have built the image on several different setups. I'm closing this issue.

closed 2019-07-18 by gwolf