Debian/raspi3-image-spec #55: Unable to boot on Rpi 3 model B

ukch, 2019-03-03:
I have tried both the [pre-built image](https://people.debian.org/~gwolf/raspberrypi3/20190206/20190206-raspberry-pi-3-buster-PREVIEW.img.xz) and building my own image using this tool. I have tried flashing using both Etcher and dd, but nothing I do seems to result in a bootable image.

What happens when I plug my Pi in is the red light comes on and the amber light flashes for about a second, then turns red. There is no HDMI output or network connectivity. Flashing a Raspbian image works as expected.

licaon-kter, 2019-03-03:
What SD card maker and model?

gwolf, 2019-03-04:
Joel Cross dijo [Sun, Mar 03, 2019 at 01:57:02PM -0800]:
> I have tried both the [pre-built
> image](https://people.debian.org/~gwolf/raspberrypi3/20190206/20190206-raspberry-pi-3-buster-PREVIEW.img.xz)
> and building my own image using this tool. I have tried flashing
> using both Etcher and dd, but nothing I do seems to result in a
> bootable image.

I will state the obvious - Did you uncompress the image before
attempting the boot? As mentioned in the Wiki:

    https://wiki.debian.org/RaspberryPi3

Once you download (and verify!) the img.gz file, you have to
uncompress it. Now, the instructions present this as a single step:

    xzcat 20190206-raspberry-pi-3-buster-PREVIEW.img.xz | dd of=/dev/sdX bs=64k oflag=dsync status=progress

But just to be sure, you can do it as two separate steps:

    xzcat 20190206-raspberry-pi-3-buster-PREVIEW.img.xz
    dd if=20190206-raspberry-pi-3-buster-PREVIEW.img of=/dev/sdX bs=64k oflag=dsync status=progress

If you do it this way, the sha256sum of the generated image is:

    3f6054b5b8e21a14f5fdf506e60e6aa305a29bcf524d3278e460866903f4a4f7

I am pointing this way because you say there is no HDMI output. This
means, not even the RPI proprietary firmware boots correctly. You
should see the traditional "rainbow screen" before Linux starts
booting.

Greetings,

ukch, 2019-03-10:
I've done a bit more testing and it seems the problem is actually with my monitor, not this image. Sorry for the false alarm!

closed 2019-03-10 by ukch

gwolf, 2019-03-11:
> Closed #55.

Thanks!