Debian/raspi3-image-spec #56: WiFi found but not working on model B

bowguy, 2019-03-09:
Using 02/06/2019 image.  Everything works except WFi.  wlan0 is found but is not ready.  Dmesg gives:

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 4.19.0-2-arm64 (debian-kernel@lists.debian.org) (gcc version 8.2.0 (Debian 8.2.0-14)) #1 SMP Debian 4.19.16-1 (2019-01-17)
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2

[   10.776637] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   10.792392] usbcore: registered new interface driver brcmfmac
[   10.826500] brcmfmac mmc1:0001:1: firmware: direct-loading firmware brcm/brcmfmac43430-sdio.bin
[   10.843859] brcmfmac mmc1:0001:1: firmware: direct-loading firmware brcm/brcmfmac43430-sdio.txt

[   10.993330] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   11.003658] vc4-drm soc:gpu: bound 3f807000.pixelvalve (ops vc4_crtc_ops [vc4])
[   11.018653] brcmfmac mmc1:0001:1: firmware: failed to load brcm/brcmfmac43430-sdio.clm_blob (-2)
[   11.048062] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
[   11.064338] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.clm_blob failed with error -2
[   11.081444] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   11.103587] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f

Thanks

mikediachenko, 2019-05-21:
The same issue here after upgrading to 4.19.42-v7+ #1219.

ungarida, 2019-05-31:
On Pi Zero W, I have the same messages and specifically `brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f` and wlan0 becomes ready but it is not working properly, meaning that most of the connections time out.

gwolf, 2019-05-31:
Davide Ungari dijo [Fri, May 31, 2019 at 01:03:05AM -0700]:
> On Pi Zero W, I have the same messages and specifically `brcmfmac:
> brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017
> 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f` and wlan0
> becomes ready but it is not working properly, meaning that most of
> the connections time out.

Interesting. Which kernel are you using? Which firmware version? My 3B
works reliably with the stock 4.19 kernel (4.19.0-2)...

ungarida, 2019-06-02:
Executing `uname -a`:
> 4.19.42+ #1219

I did some more testing and it works fine on my mobile hotspot but not on the office network. I guess in my case it is related to such network.

gwolf, 2019-06-02:
Davide Ungari dijo [Sun, Jun 02, 2019 at 03:41:09AM -0700]:
> Executing `uname -a`:
> > 4.19.42+ #1219
> 
> I did some more testing and it works fine on my mobile hotspot but
> not on the office network. I guess in my case it is related to such
> network.

Umh, that might mean it's not authenticating correctly. What are you
using to connect? Hand-generated wpa-supplicant.conf? wicd?
network-manager?

ungarida, 2019-06-02:
The network is a public office WiFi and the configuration is simply:
```
network={
   ssid="MYSSID"
   key_mgmt=NONE
}
```
The strange thing is that is working but really slowly most of the connections time out.

gwolf, 2019-07-18:
I am migrating issues (together with the project as a whole) to salsa.debian.org. Please refer there to [salsa.debian.org issue #5](https://salsa.debian.org/raspi-team/image-specs/issues/5).

closed 2019-07-18 by gwolf

bowguy, 2019-07-18:
Thanks for looking into it.  I migrated back to the standard Rasbian since they went to Buster.