Debian/raspi3-image-spec #59: CPU frequency never rises above 600 MHz under load, even with arm_freq = 1200

genpfault, 2019-04-19:
Using the SDL2 build as a benchmark I'm seeing the Buster kernel take ~2x as long to compile vs. the current Raspbian kernel:

Buster:

	$ time nice make -j 4
	real    7m44.169s
	user    27m46.941s
	sys     2m17.021s

Raspbian:

	$ time nice make -j 4
	real    3m14.458s
	user    11m50.851s
	sys     0m39.507s

SDL build procedure:

	sudo apt install cmake git-core libgbm-dev
	sudo apt build-dep libsdl2
	git clone https://github.com/SDL-mirror/SDL.git
	cd SDL
	git checkout 54cddf8952802fa94f359d21c6da46979e10ce0
	mkdir build
	cd build
	cmake ../ -DCMAKE_BUILD_TYPE=Release 
	time nice make -j 4

Buster system info:

	vcgencmd built from https://github.com/raspberrypi/userland
	
	$ uname -a
	Linux rpi3 4.19.0-4-arm64 #1 SMP Debian 4.19.28-2 (2019-03-15) aarch64 GNU/Linux
	
	$ cat /boot/firmware/cmdline.txt 
	console=tty0 console=ttyS1,115200 root=LABEL=RASPIROOT rw elevator=deadline fsck.repair=yes net.ifnames=0 cma=256M rootwait
	
	$ cat /boot/firmware/config.txt | grep -v '#'
	arm_control=0x200
	enable_uart=1
	kernel=vmlinuz-4.19.0-4-arm64
	initramfs initrd.img-4.19.0-4-arm64
	
	$ vcgencmd get_config int | grep -E 'arm_freq|core_freq|gpu_freq|sdram_freq'
	arm_freq=1200
	core_freq=250
	gpu_freq=300
	sdram_freq=450
	
	$ vcgencmd get_mem arm && vcgencmd get_mem gpu
	arm=948M
	gpu=76M
	
	# idle
	$ vcgencmd measure_clock arm && vcgencmd measure_temp
	frequency(45)=600000000
	temp=48.3'C
	
	# about 75% through the 'make'
	$ vcgencmd measure_clock arm && vcgencmd measure_temp
	frequency(45)=600000000
	temp=64.5'C

Raspbian system info:

	# version/source info:
	https://www.raspberrypi.org/downloads/raspbian/
	https://downloads.raspberrypi.org/raspbian_lite_latest
	Version: April 2019
	Release date: 2019-04-08
	Kernel version: 4.14
	SHA-256: 03ec326d45c6eb6cef848cf9a1d6c7315a9410b49a276a6b28e67a40b11fdfcf
	
	$ uname -a
	Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
	
	$ cat /boot/cmdline.txt
	dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=0db4cca0-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
	
	$ cat /boot/config.txt | grep -v '#'
	dtparam=audio=on
	
	$ vcgencmd get_config int | grep -E 'arm_freq|core_freq|gpu_freq|sdram_freq'
	arm_freq=1200
	core_freq=400
	gpu_freq=300
	sdram_freq=450
	
	$ vcgencmd get_mem arm && vcgencmd get_mem gpu
	arm=948M
	gpu=76M
	
	# idle
	$ vcgencmd measure_clock arm && vcgencmd measure_temp
	frequency(45)=600000000
	temp=46.2'C
	
	# about 75% through the 'make'
	$ vcgencmd measure_clock arm && vcgencmd measure_temp
	frequency(45)=1200000000
	temp=76.8'C

For Buster the firmware reports the `arm_freq = 1200` config but the actual `measure_clock arm` frequency never goes above 600 MHz, even under load.

licaon-kter, 2019-04-19:
Change governor from `powersave` to `ondemand`

genpfault, 2019-04-19:
@licaon-kter: `cpupower` kicks back an error:

```
root@rpi3:~# cpupower frequency-set -g ondemand
Setting cpu: 0
Error setting new values. Common errors:
- Do you have proper administration rights? (super-user?)
- Is the governor you requested available and modprobed?
- Trying to set an invalid policy?
- Trying to set a specific frequency, but userspace governor is not available,
   for example because of hardware which cannot be set to a specific frequency
   or because the userspace governor isn't loaded?
```

The Buster kernel seems to be missing the `cpufreq` sysfs entries:

```
root@rpi3:~# ls -alh /sys/devices/system/cpu/cpu*/cpufreq/*
ls: cannot access '/sys/devices/system/cpu/cpu*/cpufreq/*': No such file or directory
```

mbien, 2019-04-19:
the mainline kernel does not have a cpufreq implementation for the raspberry pi. This will cause the Pi3 to stick to idle clock if you just boot the mainline kernel (and it can't change it at runtime).

here the latest attempt i found to provide such implementation:
https://lists.infradead.org/pipermail/linux-rpi-kernel/2019-April/008634.html

you can force the hardware to stick to boost clock by setting
force_turbo=1
in config.txt. To my understanding this should not void warranty as long you don't change the voltages. You probably also want to make sure that the RPI doesn't get temperature issues if its set.

gwolf, 2019-04-20:
Michael Bien dijo [Fri, Apr 19, 2019 at 04:59:27PM -0700]:
> the mainline kernel does not have a cpufreq implementation for the
> raspberry pi. This will cause the Pi3 to stick to idle clock if you
> just boot the mainline kernel (and it can't change it at [...]

Thanks for your insight on this issue, Michael.

I think that, by default, we should _not_ generate the images set to
force_turbo=1; the best option would be for the mainline kernel to
present the ondemand governor (Romain, could you shed some light on
that?), but in the meantime... Well, this should be at least
documented!

For that matter — _many_ things should be documented. I think the
right place for this would be in the Debian Wiki:

    https://wiki.debian.org/RaspberryPi3

So, please, documentation patches welcome!

mbien, 2019-04-20:
> I think that, by default, we should _not_ generate the images set to force_turbo=1;
I agree. I updated the post to be clearer that forcing turbo clock is just a workaround and not a solution to this issue.

Another temporary option to experiment with might be to set a higher min clock frequency or forcing turbo while setting a lower max frequency - but i haven't tested that since forcing turbo worked for my usecase.

This would lock the clock to a higher (custom) frequency as a compromise until the mainline kernel has a proper cpufreq impl.

genpfault, 2019-04-22:
Looks like Buster with `force_turbo=1` hits thermal throttling about as fast but noticeably harder than Raspbian's `cpufreq` system:

[image]

Basically ran `time nice make -j4` for the SDL build 3 times in a row while logging frequency & temperature.  Trimmed the frequency/temperature data so I had 20 seconds of idle-time lead-in & lead-out before/after the builds.

gwolf, 2019-07-18:
Hmmm... Interesting, open issue we should really track down and solve! :-P

Anyway - the project is moving to salsa.debian.org. This bug is copied over there, as [salsa.debian.org issue #7](https://salsa.debian.org/raspi-team/image-specs/issues/7).

closed 2019-07-18 by gwolf