Debian/raspi3-image-spec #61: Boot partition contains three different kernel images

pavellaminar, 2019-07-09:
After copying the image to the SD card the first (boot) partition contains three different kernel images kernel.img, kernel7.img, vmlinuz-4.19.0-5-arm64. But actually it's don't prevent a normal booting

licaon-kter, 2019-07-09:
@pavellaminar please rephrase this

SuperSandro2000, 2019-07-10:
The first parition (boot) partition of ``Image 20190628_raspberry-pi-3_buster_PREVIEW.img.xz`` contains three different kernel images, namely ``kernel.img``, ``kernel7.img`` and ``vmlinuz-4.19.0-5-arm64``.

licaon-kter, 2019-07-10:
yes, and `config.txt` instructs it to load the correct one, `vmlinuz-4.19.0-5-arm64` and `initrd`, right?

gwolf, 2019-07-18:
Hi,

I am moving the project over to salsa.debian.org, and copied this issue as [salsa.debian.org issue #8](https://salsa.debian.org/raspi-team/image-specs/issues/8). Please refer to it for further information.

closed 2019-07-18 by gwolf